ros/filters #10: Jade Release

wjwwood, 2015-04-22:
@jonbinney did you mention you could release this? :wink:

jonbinney, 2015-04-22:
Suurrreee.... I just compiled under jade and all tests passed. Will release nowish

jonbinney, 2015-04-22:
Released. rosdistro PR: https://github.com/ros/rosdistro/pull/8040

Will close this issue after everything builds on the farm.

jonbinney, 2015-04-23:
It looks like the buildfarm still hasn't built this: http://www.ros.org/debbuild/jade.html

@wjwwood is the buildfarm going slow for some reason?

tfoote, 2015-04-23:
The farm's quite busy right now, we released several low level packages into jade, including catkin, so there's over 800 jobs in the queue. 

Additionally as this is a first time release it will not run until the job is generated for it when the farm reconfigures overnight. So it's going to be a while. Hopefully some time tomorrow.

jonbinney, 2015-04-23:
Is there a page that shows the build queue for the jade buildfarm? The only link I can find for the jade buildfarm is http://www.ros.org/debbuild/jade.html which doesn't give a sense of how many jobs are pending.

wjwwood, 2015-04-23:
http://jenkins.ros.org

Left hand column.

jonbinney, 2015-04-24:
Built and is now in shadow fixed.

closed 2015-04-24 by jonbinney