ros/filters #11: filters source not included in rosinstall_generator --upstream-development

tkruse, 2015-07-24:
To reproduce:
```
$ rosinstall_generator desktop --rosdistro jade --deps --wet-only --upstream-development | grep filters
The following repositories with an unknown upstream will be ignored: filters, metapackages, rosbag_migration_rule
```

The warning message hints at unknown upstream.

closed 2015-07-30 by tfoote

tkruse, 2015-07-30:
The same applies to driver_common, laser_filters, metapackages, rosbag_migration_rule, stage (when using desktop full you see the warnings). I created issues for driver_common, laser_filters, stage, as they break the catkin_make step that follows.

I don't know how urgently metapackages and rosbag_migration_rule also need to be fixed.

jonbinney, 2015-08-05:
I'm looking into fixing this now

jonbinney, 2015-08-05:
looks like this one was already fixed (didn't notice the issue was closed) I'm fising the equivalent issue on laser_filters.