ros/filters #28: Move to ros_control project?

davetcoleman, 2019-07-22:
I wasn't aware of this package until @gavanderhoorn pointed it out just now, but maybe this should live at https://github.com/ros-controls/ ? Thoughts @bmagyar?

bmagyar, 2019-07-23:
First time to see this package for me too! 
I'd be up for hosting this in that org, no problem. It seems rather low-bandwidth at the moment, are you guys planning on extending it?

tfoote, 2019-08-02:
It's currently in basic maintenance mode but would be happy to see it extended with new filters etc. There are lots of users outside of controls for example data processing. 

It might make more sense to create a downstream package `control_filters` that provides specific  filters for controls applications as plugins that can be used in the filter chains provided by this package.

davetcoleman, 2019-08-14:
I think moving it to ``ros-control`` would provide a lot more visibility and usage, and prevent potential duplicate effort, but I have no immediate needs or use cases.

peci1, 2020-06-24:
I use these filters to filter pointclouds/laserscans (via laser_filters package). It wouldn't make sense to me to depend on something from the ros-control repo.

jonbinney, 2021-07-02:
@peci1 i think this is about moving to the ros-control org, not the ros_control repo. @davetcoleman is that right?

peci1, 2021-07-02:
You're right, I got that wrong. But it still doesn't make sense to me as I treat filters as a generally usable package.

By the way, I recently released https://github.com/ctu-vras/sensor_filters which contains a set of nodes/nodelets ready-made for running filter chains on messages from sensor_msgs.

bmagyar, 2021-07-02:
Would still be happy to host it at @ros-controls, we are planning some controllers for ros2 with these guys

DLu, 2024-01-13:
I think it should stay here.