ros/filters #47: [ros2] RCLCPP parameter API change

doisyg, 2021-03-19:
I believe the recent https://github.com/ros2/rclcpp/pull/1522 breaks filters at runtime
When running `FilterChain::configure` I get:
```
scan_to_scan_filter_chain-1] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterTypeException'
[scan_to_scan_filter_chain-1]   what():  parameter 'filter1.name' has invalid type: cannot declare a statically typed parameter with an uninitialized value

```

Patrick-Lascombe, 2021-07-12:
This problem still occurs for us on the main ros2 branch, does anyone managed to replicate this issue ?

jonbinney, 2021-07-12:
I think I'm seeing this when running the chain_test on galactic. I'll try out https://github.com/ros/filters/pull/48

jonbinney, 2021-07-12:
I've merged https://github.com/ros/filters/pull/48 - it resolved the issue for me.

jonbinney, 2021-07-14:
This ended up requiring creating a separate branch for foxy. Should be fixed now in source, and releases have been created on f, g, and r.

closed 2021-07-14 by jonbinney