ros/filters #62: Regarding Read Only Setting in getParamImpl()

BruceChanJianLe, 2022-07-07:
Hi everyone, I am trying to implement "Parameters On Set Parameter Callback" feature, which I can update the parameters through a callback function. However, it seems like if I use the GetParam method, the declared parameter is set to a read only state.

Is there a special reason why it is set to read only? 

It is set at this line here. (https://github.com/ros/filters/blob/5507bc5175cb74e825ee20b9dc19e67ef6a4a022/include/filters/filter_base.hpp#L134)

jonbinney, 2025-04-18:
Sorry for not responding to this earlier, but this should get resolved through https://github.com/ros/filters/pull/83

BruceChanJianLe, 2025-04-19:
Hey @jonbinney thanks for getting back!
I had worked around this issue since then but glad to see that someone has finally taken a look into this matter.

jonbinney, 2025-04-19:
The fix isn't merged yet, so I'll leave this open until it is :-)