ros/filters #72: Release 2.1.1 to rolling

jonbinney, 2024-03-02:


jonbinney, 2024-03-02:
@tfoote @Yadunund it looks like although i have write access to the filters repo, I don't have write access to the [release repo](https://github.com/ros2-gbp/filters-release). Are you ok with giving me write access so I can do releases?

jonbinney, 2024-03-03:
Actually looks like i had write access a few years ago, it must have been removed?

tfoote, 2024-03-03:
Which repo do you need access to? If you can email me I'll get to it when I'm back to a computer.

jonbinney, 2024-03-06:
This release is blocked until the new filters release team gets added: https://github.com/ros2-gbp/ros2-gbp-github-org/issues/465

jonbinney, 2024-03-12:
Release created: https://github.com/ros/rosdistro/pull/40175

clalancette, 2024-03-19:
Closing this one, since the release was done.

closed 2024-03-19 by clalancette