ros/filters #75: Trying to build using colcon build

Ayush1285, 2024-03-29:
I am trying to build this using `colcon build` command. But it is not working, following error shows up:
`--- stderr: filters                         
CMake Error at CMakeLists.txt:26 (add_library):
  add_library INTERFACE library requires no source arguments.

CMake Error at CMakeLists.txt:31 (target_link_libraries):
  Cannot specify link libraries for target "realtime_circular_buffer" which
  is not built by this project.

---
Failed   <<< filters [6.26s, exited with code 1]`

It says INTERFACE library requires no source arguments. how can I resolve this without specifying the source files?

jonbinney, 2024-03-29:
What branch are you on? What ROS distro? What OS version?

Ayush1285, 2024-03-29:
My bad for not mentioning all the details.
Branch - ros2
ROS Distro - humble (source installation)
OS version - Ubuntu 20.04

jonbinney, 2024-03-29:
Let me try to reproduce this - I've got a similar setup

jonbinney, 2024-03-29:
This doesn't happen on 22.04 on humble; I'm guessing a recent commit added something that doesn't work with the older cmake version. Could you try a previous commit of filters using `git checkout 7b1b525c3529c00cdbe7edcb36cc53b56781df2d` and let me know if that builds?  Also, what is the output of `cmake --version`?

jonbinney, 2024-03-29:
It looks like interface libraries with sources was added in 3.19. CI for humble runs on 22.04, so I missed this problem. The interface library was added in https://github.com/ros/filters/pull/70

@wep21 @ryanf55 it looks like the interface library for realtime_circular buffer doesn't work with the cmake version on 20.04, which is a supported platform for humble. Thoughts? I could branch for humble, rollback the interface library changes, and re-release, but is there a way to support the older cmake version instead?

jonbinney, 2024-03-29:
Ubuntu 20.04 is only "recommended" to support for humble packages, but since filters is pretty core I think we should support it. Here are the cmake versions for each platform of humble:

[image]

Since interface libraries with source arguments was only added in cmake 3.19, it would seem to break support for macos and bullseye as well.

Ayush1285, 2024-03-29:
Thanks for the help!!
I'll use the previous version.

Ryanf55, 2024-03-29:
> Ubuntu 20.04 is only "recommended" to support for humble packages, but since filters is pretty core I think we should support it. Here are the cmake versions for each platform of humble: [...]

Only bullseye, true. MacOS is rolling release - it's currently 3.29. The version in REP-2000 is just what it had when they wrote the REP.
https://formulae.brew.sh/formula/cmake

Ryanf55, 2024-04-01:
Proposed patch:
https://github.com/ros/filters/pull/76

jonbinney, 2024-08-18:
I believe this was fixed by #76 .

closed 2024-08-18 by jonbinney