ros/filters #82: filter_base.hpp an error in armhf

yuqingli05, 2025-03-21:
/opt/ros/ws_moveit2/src/filters/include/filters/filter_base.hpp:219:8: error: ‘bool filters::FilterBase<T>::getParam(const std::string&, size_t&)’ cannot be overloaded with ‘bool filters::FilterBase<T>::getParam(const std::string&, unsigned int&)’
  219 |   bool getParam(const std::string & name, size_t & value)

It should be caused by the same 32-bit CPU size_t and unsigned in

jonbinney, 2025-03-21:
Out of curiosity, what 32 bit processor are you using? Even raspberry pi's are 64 bit these days and should work with arm64 instead of armhf.

Here's the line its talking about: https://github.com/ros/filters/blob/ros2/include/filters/filter_base.hpp#L219

The error makes sense..... I _think_ the right solution is to change `size_t` to `unsigned long int` or `uint64_t` So that there is one overload for each size. Then if it is called with size_t the compiler will choose the right one to call based on the architecture. I'm leaning towards `uint64_t` since that should be the underlying type, whereas `unsigned long int` could vary.

jonbinney, 2025-03-21:
Want to change it to uint64_t and make a PR? (i don't have a 32 bit system to test on right now)

yuqingli05, 2025-03-23:
> Want to change it to uint64_t and make a PR? (i don't have a 32 bit system to test on right now)

I compiled it using qemu+docker+ubuntu24.04 because I want to run it on an embedded CPU (A7).
This error is because unsigned int and size_t  are exactly the same on a 32-bit CPU, resulting in duplicate function definitions.
After deleting https://github.com/ros/filters/blob/ros2/include/filters/filter_base.hpp#L201, it compiled normally.